ObKo/stm32-cmake #268: HAL drivers detection without knowing the list of expected drivers

atsju, 2021-08-12:
**Is your feature request related to a problem? Please describe.**
as shown in #255 update of HAL could lead to new modules that are not automatically managed by the repo. 
Moreover adding them could cause issues with older versions of HAL repos.

**Describe the solution you'd like**
HAL files should be discovered automatically and not depending on HAL_DRIVERS_XX HAL_LL_DRIVERS_XX HAL_EX_DRIVERS_XX

Hish15, 2021-08-16:
I love the idea! This is 1) doable and 2) makes more sense!
If you are not already working on this @atsju I might give it a try :)

atsju, 2021-08-16:
Hello there. I am the one opening the ticket my friend :)

atsju, 2021-08-16:
Go ahead. I didn't start anything on this.

PR #302 (2022-11-04): HAL SCAN 2
This does a lot of changes, but I did clean commits.
I suggest reviewing commit by commit.

Previously some tests selected all families on find_package command by setting no family.
Then the test proceeded to downloading the HAL for only one family. This is no longer an option since the HAL files are needed to retrieve the component list for each family. I changed the tests to test feed only the tested family to the find_package command.

close #268

Tested on

- [x] STM32F4
- [x] STM32H7 
- [x] STM32L0
closed 2022-11-04 by Hish15