ObKo/stm32-cmake #269: How to build examples/blinky

zzs213, 2021-08-12:
I'm a beginner of stm32. So try to build a simple example.

I'm using a docker image as work env, Which with arm toolchain gcc-arm-none-eabi-10.3-2021.07-x86_64-linux.tar.bz2 installed.

I try the flowwing commands

```sh
export STM32_TOOLCHAIN_PATH="/tools/gcc-arm-none-eabi-10.3-2021.07"
export TARGET_TRIPLET=arm-none-eabi

cd examples/blinky
mkdir build
cd build
cmake ..
```

But Error as flowwing:

> -- Detected toolchain path STM32_TOOLCHAIN_PATH in environmental variables:
> -- /tools/gcc-arm-none-eabi-10.3-2021.07
> -- No STM32_TOOLCHAIN_PATH specified, using default: /usr
> -- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
> -- The C compiler identification is GNU 10.3.1
> -- The ASM compiler identification is GNU
> -- Found assembler: /tools/gcc-arm-none-eabi-10.3-2021.07/bin/arm-none-eabi-gcc
> -- Check for working C compiler: /tools/gcc-arm-none-eabi-10.3-2021.07/bin/arm-none-eabi-gcc
> -- Check for working C compiler: /tools/gcc-arm-none-eabi-10.3-2021.07/bin/arm-none-eabi-gcc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Search for CMSIS families: STM32F4
> -- Search for CMSIS RTOS: RTOS;RTOS_V2
> -- Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default STM32_CUBE_F4_PATH: /opt/STM32CubeF4
> CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
>   Could NOT find CMSIS (missing: CMSIS_INCLUDE_DIRS CMSIS_SOURCES STM32F4)
>   (found version "v1.1.0")
> Call Stack (most recent call first):
>   /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
>   /home/worker/roots/designs/play-stm32/stm32-cmake/cmake/FindCMSIS.cmake:296 (find_package_handle_standard_args)
>   CMakeLists.txt:30 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/worker/roots/designs/play-stm32/stm32-cmake/examples/blinky/build/CMakeFiles/CMakeOutput.log".
> See also "/home/worker/roots/designs/play-stm32/stm32-cmake/examples/blinky/build/CMakeFiles/CMakeError.log".

As README.md said:  `STM32_CUBE_<FAMILY>_PATH` should be set.

But I'm not known which package should be installed for STM32_CUBE_xxx.

Can somebody help me?

atsju, 2021-08-12:
warning is :
` Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default STM32_CUBE_F4_PATH`

you need to clone this : https://github.com/STMicroelectronics/STM32CubeF4
and set `STM32_CUBE_F4_PATH ` then it will work

atsju, 2021-08-12:
info for #246

closed 2021-08-12 by zzs213