ObKo/stm32-cmake #274: Blinky example won't start when changing blinky.c -> blinky.cpp

Osteri, 2021-11-07:
**Describe the bug**
C++ version of main will compile fine, but LED won't start flashing.

**Sources to reproduce**
I think it's so simple I won't provide a fork, but I can if it's really needed.

**Steps to reproduce**
1. Go to Blinky example project directory
2. Rename blinky.c -> blinky.cpp
2. Change the source file in CMakeLists.txt
3. Add CXX as the project language in CMakeLists.txt
4. Compile and flash your microcontroller (I'm using USB DFU if that makes a difference)
5. LED is not flashing as it should

**Expected behavior**
LED should flash as in the C example. Nothing will happen with C++ version. Something wrong with startup procedure on C++?

**Environment (please complete the following information):**
 - OS: Fedora 34
 - Compiler: arm-none-eabi-gcc (Fedora 10.2.0-4.fc34) 10.2.0
 - stm32-cmake: commit 12240e0
 - cmake: 3.19.7
 - HAL/cube/CMSIS: 3d7a3c1 Release v2.6.6, 313c279 Release v1.7.12, 96d6da4 Official ARM version: v5.6.0

Osteri, 2021-11-07:
If I'll add `STM32::Nano` target for linker (as guided in the docs), it will give me linker errors:
```
[ 87%] Linking CXX executable executable.elf
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(lib_a-abort.o): in function `abort':
/builddir/build/BUILD/newlib-4.1.0/build-nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/libc/stdlib/../../../../../../../../newlib/libc/stdlib/abort.c:59: undefined reference to `_exit'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(lib_a-signalr.o): in function `_kill_r':
/builddir/build/BUILD/newlib-4.1.0/build-nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:53: undefined reference to `_kill'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(lib_a-signalr.o): in function `_getpid_r':
/builddir/build/BUILD/newlib-4.1.0/build-nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:83: undefined reference to `_getpid'
collect2: error: ld returned 1 exit status
```

Hish15, 2021-11-08:
Hello @Osteri,

I will give it a try later on, but building on your use of STM32::Nano :
Using this, your are expected to define a bunch of syscall functions.
Can you try to link against STM32::Nano AND STM32::NoSys and see if it fixes it?

atsju, 2021-11-08:
I think there are missing some `extern C`.
Most likelly there is a problem at linking where `void SysTick_Handler(void)` from the main.cpp is not called.
See here: [stackoverflow: what-is-the-effect-of-extern-c-in-c](https://stackoverflow.com/questions/1041866/what-is-the-effect-of-extern-c-in-c)

This is a very simple/minimalist C example. Converting to C++ will need slightly more than changing name to .cpp to work correclty but it can definitelly be done. It's likelly you will need to create a blinky.h file to link correclty between C and C++.

Hish15, 2021-11-08:
Ok I did some testing.
The problem is indead the `void SysTick_Handler(void)` this does not have the same behavior when defined in a C or C++ file (see @atsju  link).
The fix is really simple, say explicitly that you want to define the function Systick_Handler without any name mangling using this  in your your blincky.cpp file.
```
extern "C" void SysTick_Handler(void)
```

ST made their part to allow you to use their function direcly in C++ (The .h file have an extern "C" if called from a C++ build unit). So this is the only change required in your case.

Osteri, 2021-11-08:
Thank you, it worked indeed. Awesome project btw! 

Would you mind if I make a pull request with `blinky-cpp` project as a separate C++ example? Ping @Hish15 @atsju 

I think it's a very common and useful pattern to write all of your business logic in C++ (including main.cpp) and then `extern` away all the C libraries. This is what we do and it's pretty nice as the GCC ARM already supports modern C++14 standard.

Hish15, 2021-11-08:
That's a good news @Osteri!

Do the PR yes, maybe it's a good idea not to include the C function in the main.cpp file in this PR?

I let you make the PR so we can discuss implementation details there.
GCC ARM let you do more than C++14 (I use C++17 on a GCC 7 version ^^))

PR #285 (2022-02-19): Add blinky C++ example
Replace PR close #275 and issue close #274.
closed 2021-11-08 by Osteri