ObKo/stm32-cmake #276: common.cmake get_filename_component(STM32_TOOLCHAIN_PATH ${STM32_TOOLCHAIN_PATH} DIRECTORY) wrong behavior?

andyinno, 2021-11-17:
**Describe the bug**
If I do not explicity set STM32_TOOLCHAIN_PATH cmake exit with this error:
```bash
Not searching for unused variables given on the command line.
[cmake] CMake Error at Modules/stm32_cmake/cmake/stm32/common.cmake:29 (get_filename_component):
[cmake]   get_filename_component called with incorrect number of arguments
[cmake] Call Stack (most recent call first):
[cmake]   Modules/stm32_cmake/cmake/stm32_gcc.cmake:4 (include)
[cmake]   /usr/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
[cmake]   CMakeLists.txt:4 (project)
```
**Sources to reproduce**

I am just using vscode with the compiler in my path and with the following parameters that are passed to cmake:
```json
  "cmake.configureArgs": [
        "-DSTM32_CUBE_F4_PATH=${workspaceFolder}/Src/ST/STM32CubeF4",
        "-DFREERTOS_PATH=${workspaceFolder}/Src/Application/Middlewares/Third_Party/FreeRTOS/"
    ],
```
the output from the initialization is:

`Executing command: /usr/bin/cmake --no-warn-unused-cli -DSTM32_CUBE_F4_PATH=/workspaces/KPIO06CTRL/Src/ST/STM32CubeF4 -DFREERTOS_PATH=/workspaces/KPIO06CTRL/Src/Application/Middlewares/Third_Party/FreeRTOS/ -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=arm-none-eabi-gcc -DCMAKE_TOOLCHAIN_FILE:FILEPATH=/workspaces/KPIO06CTRL/Src/Modules/stm32_cmake/cmake/stm32_gcc.cmake -H/workspaces/KPIO06CTRL/Src -B/workspaces/KPIO06CTRL/build -G Ninja
`

**Expected behavior**
That cmake configures correctly
 help explain your problem.

**Environment (please complete the following information):**
 - OS: Ubuntu
 - stm32-cmake: commit 2dc0456
 - cmake: 3.18
 - HAL/cube/CMSIS: v1.26.1

**Additional context**
If I remove the line with the get_filename_components the compiler is correctly detected and the compilation goes on.

In my opinion this line should not be there, if I do not set the STM32_TOOLCHAIN_PATH, how could this variable be used in the source file?

andyinno, 2021-11-17:
Sorry, I hit the comment button before adding other context.

I had a previously used application that was not updated with the latest code in more or less 6 months, I am trying to switch branch from the "new cmake" to the current master.

I am hitting two errors, this is the first. The other one I will add a different issue.

Hish15, 2021-11-17:
Hi again,

Can we see a snapshot of your CmakeLists.txt too please?

atsju, 2021-11-17:
> Hi again,
> 
> Can we see a snapshot of your CmakeLists.txt too please?

Hi,
Moreover, are the examples working or is only your legacy code not working ?

andyinno, 2021-11-17:
Hello Again,

thank you to you both for your fast reply.

As @atsju pointed out it was correct to test the samples.

At the moment I applied my modification to the blinky-example and it seems to me that I am observing the exact same behavior.

As anticipated, for the development I am using a devcontainer but the same I can see outside the container.

cmake-kits.json configured as:

```json
[
    {
        "name": "ST GCC",
        "keep": true,
        "toolchainFile": "${workspaceRoot}/../../cmake/stm32_gcc.cmake",
        "compilers": {
            "C": "arm-none-eabi-gcc"
        }
    }
]
```

settings.json configured as

```json
{
    "cmake.sourceDirectory": "${workspaceFolder}/.",
    "cmake.configureArgs": [
        "-DSTM32_CUBE_F4_PATH=${workspaceFolder}/ST/STM32CubeF4",
        "-DFREERTOS_PATH=${workspaceFolder}/Src/Application/Middlewares/Third_Party/FreeRTOS/"
    ],
}

```

CMakeLists.txt file:

```cmake
cmake_minimum_required(VERSION 3.16)
#set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/../../cmake/stm32_gcc.cmake)

project(stm32-blinky C ASM)
set(CMAKE_INCLUDE_CURRENT_DIR TRUE)

# Configure here which STM32 target(s) to build
option(BLINKY_F4_EXAMPLE "Compile F4 example" ON)
option(BLINKY_F1_EXAMPLE "Compile F1 example" OFF)
option(BLINKY_L0_EXAMPLE "Compile L0 example" OFF)

set(HAL_COMP_LIST RCC GPIO CORTEX)
set(CMSIS_COMP_LIST "")

if(BLINKY_F4_EXAMPLE)
    list(APPEND CMSIS_COMP_LIST STM32F4)
    list(APPEND HAL_COMP_LIST STM32F4)
endif()

if(BLINKY_F1_EXAMPLE)
    list(APPEND CMSIS_COMP_LIST STM32F1)
    list(APPEND HAL_COMP_LIST STM32F1)
endif()

if(BLINKY_L0_EXAMPLE)
    list(APPEND CMSIS_COMP_LIST STM32L0)
    list(APPEND HAL_COMP_LIST STM32L0)
endif()

[... 144 lines cut ...]
[build] -- Search for CMSIS RTOS: RTOS;RTOS_V2
[build] -- Search for HAL families: STM32F4
[build] -- Search for HAL drivers: RCC;GPIO;CORTEX
[build] -- Search for HAL LL drivers: 
[build] -- Could not read the HAL version from package.xml for STM32F4
[build] -- Configuring done
[build] -- Generating done
[build] -- Build files have been written to: /home/andrea/Git/stm32-cmake/examples/blinky/build
[build] [1/10  10% :: 0.011] Generating F407VG.ld
[build] [8/10  20% :: 0.024] Building ASM object CMakeFiles/stm32-blinky-f4.dir/ST/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f407xx.s.obj
[build] [8/10  30% :: 0.082] Building C object CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj
[build] [8/10  40% :: 0.086] Building C object CMakeFiles/stm32-blinky-f4.dir/ST/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c.obj
[build] [8/10  50% :: 0.099] Building C object CMakeFiles/stm32-blinky-f4.dir/ST/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c.obj
[build] [8/10  60% :: 0.110] Building C object CMakeFiles/stm32-blinky-f4.dir/ST/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c.obj
[build] [8/10  70% :: 0.121] Building C object CMakeFiles/stm32-blinky-f4.dir/ST/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj
[build] [8/10  80% :: 0.164] Building C object CMakeFiles/stm32-blinky-f4.dir/ST/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c.obj
[build] [9/10  90% :: 0.199] Linking C executable stm32-blinky-f4.elf
[build] [10/10 100% :: 0.204] Target Sizes:
[build]    text	   data	    bss	    dec	    hex	filename
[build]    2672	   1096	   1572	   5340	   14dc	stm32-blinky-f4.elf
[build] Build finished with exit code 0

```
After taking the output I noticed that I did not copied the package.xml file but just the libraries in the correct position, therefore you can see the output `Could not read the HAL version from package.xml for STM32F4`
I rechecked the output with the file package.xml in the correct position but the result is unchanged.

Hish15, 2021-11-20:
Hi,
`VECT_TAB_OFFSET` is defined in `system_stm32${FAMILY_L}xx.c` this is found while using FindCMSIS.
It looks like you did found it while configuring cmake (otherwise you would have an error message). We can see that the file is indeed built...

As you can see here :
https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c#L104

It is indeed defined. Did you made any change to this file? Is it properly mounted into your docker container and all?

andyinno, 2021-11-20:
Hello! 

Mmm...

Can't say now. I need to check.
Anyway this issue is not related to the VECT_TAB_OFFSET macro.

I can try again with a clean setup.

What I was noticing is that if STM32_TOOLCHAIN_PATH is not defined, the configure process exits with error.

The VECT_TAB_OFFSET could be related to a change to a previous version made for adding it as a Param from my cmakelists.txt

atsju, 2022-01-29:
Hi @andyinno 
Do you have any update on this issue ? Or may we close it ?