ObKo/stm32-cmake #277: Custom linker script applied with the one defined by the module

andyinno, 2021-11-17:
**Describe the bug**
After the compilation the linker warns about a redeclaration of regions ROM, RAM, etc...

**Sources to reproduce**

I am using a custom linker script where I altered the flash size in order to fit a bootloader and the eeprom emulation library.
Therefore I needed to change the start address of my application.

Where I am setting the target I am calling the function
```cmake
stm32_add_linker_script(application PRIVATE ${CMAKE_CURRENT_BINARY_DIR}/application_linker_script.ld)
```
in order to add my linker script. If I want to remove the warning I need to comment out the line:

```cmake
#stm32_add_linker_script(CMSIS::STM32::${DEVICE}${CORE_C} INTERFACE "${OUTPUT_LD_FILE}")
```

in function cmsis_generate_default_linker_script in file FindCMSIS.cmake

From the [documentation](https://github.com/ObKo/stm32-cmake#linker-script--variables) I understand that I can completely override the linker script, am I wrong?

Hish15, 2021-11-17:
Hello,

Are you **NOT** linking against the target `CMSIS::STM32::<DEVICE>` if you do, you will have the generated linker script used.
You must link against the target `CMSIS::STM32::<TYPE>`, are you?

andyinno, 2021-11-17:
Hello, 

I am linking against `CMSIS::STM32::F412VG`. So in your opinion I should link to `CMSIS::STM32::F4`?

I made a test and yes, you are correct. Now the linker_script is not added to my target but I have a question.

Do I lose anything targetting `CMSIS::STM32::F4`?

Hish15, 2021-11-17:
You do lose things, from README.md
```
CMSIS creates the following targets:

* `CMSIS::STM32::<FAMILY>` (e.g. `CMSIS::STM32::F4`) - common includes, compiler flags and defines for family
* `CMSIS::STM32::<TYPE>` (e.g. `CMSIS::STM32::F407xx`) - common startup source for device type, depends on `CMSIS::STM32::<FAMILY>`
* `CMSIS::STM32::<DEVICE>` (e.g. `CMSIS::STM32::F407VG`) - linker script for device, depends on `CMSIS::STM32::<TYPE>`
```

Consider using `CMSIS::STM32::F412xx` in your case

andyinno, 2021-11-17:
Great, thank you for your help.

I missed that part of the documentation.
Do you think it makes sense to add to the documentation what you just explained to me?
My English is not perfect but I can provide an update to the README.

thank you again
Andrea

closed 2021-11-17 by andyinno

Hish15, 2021-11-17:
You are welcome.

The Readme.md has the information just bellow the part I quoted before...

You are not the first to have an issue will trying to use a custom linker script. I think that we need to make a FAQ section or change the way the Readme file is structured.

```
So, if you don't need linker script, you can link only CMSIS::STM32::<TYPE> library and provide your own script using stm32_add_linker_script function
```

andyinno, 2021-11-17:
Hi @Hish15 

Please forgive me if I am appending more noise to this issue.

It is unclear to me the `CMSIS::STM32::F407xx`.

If I search for it:

```bash
[cmake] CMake Error at Modules/stm32_cmake/cmake/stm32/common.cmake:103 (message):
[cmake]   Invalid/unsupported device: F412XX
```

If I search for `CMSIS::STM32::F412`:
```bash
[cmake]  Target "application" links to target "CMSIS::STM32::F412" but the
[cmake]   target was not found.  Perhaps a find_package() call is missing for an
[cmake]   IMPORTED target, or an ALIAS target is missing?

```

Also... If I check here https://github.com/ObKo/stm32-cmake/blob/master/examples/custom-linker-script/CMakeLists.txt

```cmake
cmake_minimum_required(VERSION 3.16)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/../../cmake/stm32_gcc.cmake)

project(stm32-custom-linker-script C ASM)

find_package(CMSIS COMPONENTS STM32F407VG REQUIRED)

set(PROJECT_SOURCES
    main.c
)

add_executable(stm32-custom-linker-script ${PROJECT_SOURCES})
target_link_libraries(stm32-custom-linker-script CMSIS::STM32::F407xx STM32::NoSys)
stm32_add_linker_script(stm32-custom-linker-script PRIVATE F407VG.ld)
stm32_print_size_of_target(stm32-custom-linker-script)
```

The find_package calls for STM32F407VG
and then it adds the linker script. Is not this supposed to be as my initial code?

andyinno, 2021-11-17:
Ok, the difference is here:

`find_package(CMSIS COMPONENTS STM32F407VG REQUIRED)`

with full component

`target_link_libraries(stm32-custom-linker-script CMSIS::STM32::F407xx STM32::NoSys)`

with the xx version.

But...

```bash
[cmake]   Target "application" links to target "CMSIS::STM32::F412xx" but the
[cmake]   target was not found.  Perhaps a find_package() call is missing for an
[cmake]   IMPORTED target, or an ALIAS target is missing?
```

Hish15, 2021-11-17:
My bad,
https://github.com/ObKo/stm32-cmake/blob/master/cmake/stm32/f4.cmake
F412Vx in your case :)

andyinno, 2021-11-18:
> My bad, https://github.com/ObKo/stm32-cmake/blob/master/cmake/stm32/f4.cmake F412Vx in your case :)

Thanks!
Now its clear to me where the information were