ObKo/stm32-cmake #278: Unused STM32_TOOLCHAIN_PATH environmental variable

darkmoon32, 2021-12-01:
When I set the `STM32_TOOLCHAIN_PATH` as the environmental variable, it gets ignored and the default is used instead.

Steps to reproduce:
```bash
git clone https://github.com/ObKo/stm32-cmake.git
export STM32_TOOLCHAIN_PATH="/opt/gcc-arm-none-eabi-10.3-2021.10"
cmake -S examples/fetch-cube/ -B build
cmake --build build
```

cmake output:
```
$ cmake -S examples/fetch-cube/ -B build
-- Detected toolchain path STM32_TOOLCHAIN_PATH in environmental variables: 
-- /opt/gcc-arm-none-eabi-10.3-2021.10
-- No STM32_TOOLCHAIN_PATH specified, using default: /usr
-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Populating stm32cubef4
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/stm32-cmake/build/_deps/stm32cubef4-subbuild
....
```

I have managed to solve this by reorganizing the block of code [here](https://github.com/ObKo/stm32-cmake/blob/master/cmake/stm32/common.cmake#L16) but I am not sure if I am not missing something.

atsju, 2021-12-01:
Hello,
shouldn't it be `export STM32_TOOLCHAIN_PATH="/opt/gcc-arm-none-eabi-10.3-2021.10/bin"` ?
Also do not hesitate to make a branch and push a pull request so that we can see how you modified the code. Depending on the modification we might understant if the problem come from the repo or from the usage.

darkmoon32, 2021-12-01:
Hi,
I do not think so. The common module creates path to binaries [here](https://github.com/ObKo/stm32-cmake/blob/master/cmake/stm32/common.cmake#L43). It works fine with the same path when I use `cmake -S examples/fetch-cube/ -B build -DSTM32_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-10.3-2021.10`.

My modifications are the following:
```cmake
if(NOT STM32_TOOLCHAIN_PATH)
    if(DEFINED ENV{STM32_TOOLCHAIN_PATH})
        message(STATUS "Detected toolchain path STM32_TOOLCHAIN_PATH in environmental variables: ")
        message(STATUS "$ENV{STM32_TOOLCHAIN_PATH}")
        set(STM32_TOOLCHAIN_PATH $ENV{STM32_TOOLCHAIN_PATH})
    else()
      if(NOT CMAKE_C_COMPILER)
          set(STM32_TOOLCHAIN_PATH "/usr")
          message(STATUS "No STM32_TOOLCHAIN_PATH specified, using default: " ${STM32_TOOLCHAIN_PATH})
      else()
          # keep only directory of compiler
          get_filename_component(STM32_TOOLCHAIN_PATH ${CMAKE_C_COMPILER} DIRECTORY)
          # remove the last /bin directory
          get_filename_component(STM32_TOOLCHAIN_PATH ${STM32_TOOLCHAIN_PATH} DIRECTORY)
      endif()
    endif()
    file(TO_CMAKE_PATH "${STM32_TOOLCHAIN_PATH}" STM32_TOOLCHAIN_PATH)
endif()
```

atsju, 2022-01-29:
Sorry for delay.
I think you are right and opened a PR with the proposed changes.

closed 2022-04-13 by darkmoon32