ObKo/stm32-cmake #280: Weird --sysroot argument escaping on Windows

Inujel, 2022-01-26:
When building on Windows (with Ninja), the sysroot argument is weirdly escaped:
```
...
[build] -- Configuring done
[build] -- Generating done
[build] -- Build files have been written to: F:/Pro/rd-46/stm32-cmake/examples/blinky/build
[build] [7/8  12% :: 0.054] Building C object CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj
[build] FAILED: CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj 
[build] F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools\bin\arm-none-eabi-gcc.exe -DSTM32F4 -DSTM32F407xx -IF:/Pro/rd-46/stm32-cmake/examples/blinky/build -IF:/Pro/rd-46/stm32-cmake/examples/blinky -isystem C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Inc -isystem C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/CMSIS/Include -isystem C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/CMSIS/Device/ST/STM32F4xx/Include -g --sysroot=\"F:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346/tools/arm-none-eabi\" -mthumb -mabi=aapcs -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin -ffast-math -Og -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --specs=nosys.specs -MD -MT CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj -MF CMakeFiles\stm32-blinky-f4.dir\blinky.c.obj.d -o CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj -c F:/Pro/rd-46/stm32-cmake/examples/blinky/blinky.c
[build] cc1.exe: error: "F:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346/tools/arm-none-eabi"/build/gnu-tools-for-stm32_9-2020-q2-update.20201001-1621/install-mingw/lib/gcc/arm-none-eabi/9.3.1/../../../../include: Invalid argument
```

Using ```"--sysroot=${TOOLCHAIN_SYSROOT}"``` rather than ```--sysroot="${TOOLCHAIN_SYSROOT}"``` in [utilities.cmake](https://github.com/ObKo/stm32-cmake/blob/2dc0456bc5317a467ea25fc4d36dbc8e2178bdee/cmake/stm32/utilities.cmake#L20) seems to fix the issue.

atsju, 2022-01-26:
Please be more precise about how you build and how to reproduce. I build on windows with Ninja and have no issue. 
If we cannot reproduce we cannot integrate the proposed fix.

Inujel, 2022-01-26:
I am using cmake 3.22.1, ninja 1.10.2 and the toolchain from STM32CubeIDE 1.6.1. 
I'm trying to build the blinky sample.

CMake is invoked in Powershell with 
```
cmake .. -GNinja -DSTM32_TOOLCHAIN_PATH=F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools -DSTM32_CUBE_F4_PATH=C:\Users\julie\STM32Cube\Repository\STM32Cube_FW_F4_V1.26.2 -DSTM32_TARGET_TRIPLET=arm-none-eabi
```

Here's a complete log:
```
PS F:\Pro\rd-46\stm32-cmake\examples\blinky\build> cmake .. -GNinja -DSTM32_TOOLCHAIN_PATH=F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools -DSTM32_CUBE_F4_PATH=C:\Users\julie\STM32Cube\Repository\STM32Cube_FW_F4_V1.26.2 -DSTM32_TARGET_TRIPLET=arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: F:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346/tools/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: F:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346/tools/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Search for CMSIS families: STM32F4
-- Search for CMSIS RTOS: RTOS;RTOS_V2
-- Found CMSIS: C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/CMSIS/Include;C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/CMSIS/Device/ST/STM32F4xx/Include;C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS;C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 (found version "5.4.0") found components: STM32F4
-- Search for HAL families: STM32F4
-- Search for HAL drivers: RCC;GPIO;CORTEX
-- Search for HAL LL drivers:
-- Found HAL: C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Inc  found components: RCC GPIO CORTEX STM32F4
-- Configuring done
-- Generating done
-- Build files have been written to: F:/Pro/rd-46/stm32-cmake/examples/blinky/build
PS F:\Pro\rd-46\stm32-cmake\examples\blinky\build> cmake --build .
[2/10] Building C object CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj
FAILED: CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj
F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools\bin\arm-none-eabi-gcc.exe -DSTM32F4 -DSTM32F407xx -IF:/Pro/rd-46/stm32-cmake/examples/blinky/build -IF:/Pro/rd-46/stm32-cmake/examples/blinky -isystem C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Inc -isystem C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/CMSIS/Include -isystem C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/CMSIS/Device/ST/STM32F4xx/Include --sysroot=\"F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools/arm-none-eabi\" -mthumb -Wall -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --specs=nosys.specs -MD -MT CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj -MF CMakeFiles\stm32-blinky-f4.dir\C_\Users\julie\STM32Cube\Repository\STM32Cube_FW_F4_V1.26.2\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_cortex.c.obj.d -o CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj -c C:/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c
cc1.exe: error: "F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools/arm-none-eabi"/build/gnu-tools-for-stm32_9-2020-q2-update.20201001-1621/install-mingw/lib/gcc/arm-none-eabi/9.3.1/../../../../include: Invalid argument
cc1.exe: error: "F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools/arm-none-eabi"/usr/include: Invalid argument
[3/10] Building C object CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c.obj
FAILED: CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c.obj
F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools\bin\arm-none-eabi-gcc.exe -DSTM32F4 -DSTM32F407xx -IF:/Pro/rd-46/stm32-cmake/examples/blinky/build -IF:/Pro/rd-46/stm32-cmake/exampl [... cut ...] -none-eabi/9.3.1/../../../../include: Invalid argument
cc1.exe: error: "F:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools/arm-none-eabi"/usr/include: Invalid argument
[8/10] Building ASM object CMakeFiles/stm32-blinky-f4.dir/C_/Users/julie/STM32Cube/Reposito...F4_V1.26.2/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f407xx.s.obj
ninja: build stopped: subcommand failed.
```

atsju, 2022-01-27:
Thank you for details.

@Hish15  I think we should correct the path handling in a CMAKE way https://cmake.org/cmake/help/latest/command/file.html
But do we use 3.20 features like https://cmake.org/cmake/help/latest/command/cmake_path.html#cmake-path-to-native-path-list or the old one ?

Hish15, 2022-01-27:
I think we should stick with 3.16, but I agree with the proper use of "file".

atsju, 2022-01-29:
Hi @Inujel, 
I have been unable to reproduce your issue with what I would call vanillia gcc from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads).
```
PS C:\GIT\stm32-cmake\examples\blinky\build> cmake .. -GNinja -DSTM32_TOOLCHAIN_PATH=C:\10-2020-q4-major -DSTM32_CUBE_F4_PATH=C:\GIT\STM32CubeF4 -DSTM32_TARGET_TRIPLET=arm-none-eabi
-- The C compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/10-2020-q4-major/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/10-2020-q4-major/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Search for CMSIS families: STM32F4
-- Search for CMSIS RTOS: RTOS;RTOS_V2
-- Found CMSIS: C:/GIT/STM32CubeF4/Drivers/CMSIS/Include;C:/GIT/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Include;C:/GIT/STM32CubeF4/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS;C:/GIT/STM32CubeF4/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 (found version "5.4.0") found components: STM32F4
-- Search for HAL families: STM32F4
-- Search for HAL drivers: RCC;GPIO;CORTEX
-- Search for HAL LL drivers:
-- Found HAL: C:/GIT/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Inc  found components: RCC GPIO CORTEX STM32F4
-- Configuring done
-- Generating done
-- Build files have been written to: C:/GIT/stm32-cmake/examples/blinky/build
PS C:\GIT\stm32-cmake\examples\blinky\build> cmake --build .
[10/10] Target Sizes:
   text    data     bss     dec     hex filename
   2664    1096    1572    5332    14d4 stm32-blinky-f4.elf
PS C:\GIT\stm32-cmake\examples\blinky\build> cmake --version
cmake version 3.22.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
PS C:\GIT\stm32-cmake\examples\blinky\build> ninja --version
1.10.2
PS C:\GIT\stm32-cmake\examples\blinky\build>
```

But then I installed your toolchain and have been able to reproduce the issue.
However, I have been able to solve by simply adding `/bin` to your `STM32_TOOLCHAIN_PATH`

```
PS C:\GIT\stm32-cmake\examples\blinky\build> cmake .. -GNinja -DSTM32_TOOLCHAIN_PATH=C:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\bin -DSTM32_CUBE_F4_PATH=C:\GIT\STM32CubeF4 -DSTM32_TARGET_TRIPLET=arm-none-eabi
-- The C compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Search for CMSIS families: STM32F4
-- Search for CMSIS RTOS: RTOS;RTOS_V2
-- Found CMSIS: C:/GIT/STM32CubeF4/Drivers/CMSIS/Include;C:/GIT/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Include;C:/GIT/STM32CubeF4/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS;C:/GIT/STM32CubeF4/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 (found version "5.4.0") found components: STM32F4
-- Search for HAL families: STM32F4
-- Search for HAL drivers: RCC;GPIO;CORTEX
-- Search for HAL LL drivers:
-- Found HAL: C:/GIT/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Inc  found components: RCC GPIO CORTEX STM32F4
-- Configuring done
-- Generating done
-- Build files have been written to: C:/GIT/stm32-cmake/examples/blinky/build
PS C:\GIT\stm32-cmake\examples\blinky\build> cmake --build .
[10/10] Target Sizes:
   text    data     bss     dec     hex filename
   2664    1096    1572    5332    14d4 stm32-blinky-f4.elf
```

Considering that documentation is stating 

> STM32_TOOLCHAIN_PATH - where toolchain is located, default: /usr. Alternatively you can add the folder containing the toolchain binaries to the system path

I would just close this issue but I tried to understand a bit better and looking specifically at your error:
`5.0.202011040924/tools/arm-none-eabi"/usr/include: Invalid argument` but there is no such thing as `/usr/include` in any cmake files. I also rebuilded with [official 9-2020-q2-update toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) and this one works fine.
Comparing the 2 toolchains it seems that the ST one has been rebuilt and is different (maybe partially for linux).

**As a conclusion do not use the ST toolchain which has been wrapped for STM32cubeIDE.**

closed 2022-01-29 by atsju

Inujel, 2022-01-29:
Ok thank you very much for your time!