ObKo/stm32-cmake #286: Update MP1 cube version to 1.6.0 when delivered

atsju, 2022-02-22:
see https://github.com/STMicroelectronics/STM32CubeMP1/pull/13

and modify [tests/cmsis/CMakeLists.txt](https://github.com/ObKo/stm32-cmake/pull/283/files#diff-c143a906280b91d7e0c179e631912348f6e35e396cb07acff6af990fe3a2bb55) to remove
```cmake
if("MP1" IN_LIST TEST_FAMILIES)
    message(WARNING "CMSIS for MP1 devices requires HAL, this test is expected to fail for the MP1 family")
endif()
```
discoverd in #238 

mention @Inujel

atsju, 2022-07-12:
Will also need https://github.com/STMicroelectronics/STM32CubeMP1/pull/16

Hish15, 2023-01-02:
@atsju, what are the news on this ?

atsju, 2023-01-03:
ST made half of needed changes. Second mentioned PR is still needed. I suppose ST will review open PRs before next release. Let's wait.