ObKo/stm32-cmake #292: Can't compile on windows

MarkMan0, 2022-06-13:
**Describe the bug**

I can't compile any of the examples on windows

**Steps to reproduce**

I've put these lines into the CMakeLists.txt, right after `cmake_minimum_required`.

```
cmake_minimum_required(VERSION 3.16)

set(STM32_TOOLCHAIN_PATH "C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/11.2 2022.02")
set(STM32_TARGET_TRIPLET arm-none-eabi)
set(STM32_CUBE_F4_PATH "C:/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0")
```

I'm compiling in Visual Studio Code, using the Cmake extension, or manually from the command line. Both seem to produce the same error. For generator/kit, I chose mingw-w64.

Command do configure:
```
cmake -S . -B build/ -DSTM32_TOOLCHAIN_PATH="C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/11.2 2022.02" -DSTM32_TARGET_TRIPLET="arm-none-eabi" -DSTM32_CUBE_F4_PATH="C:/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0"
```

Command do compile:
```
cmake --build build/ --config Release --target stm32-blinky-f4 -j 14 --
```

These commands work on a different, windows program using cmake.

I've tried to define the three variables in the command, and in the CMakeLists.txt file too, but same error.

Configuration step runs without error, but when I try to compile, I get the following error: 
```
$ cmake --build build/ --config Release --target stm32-blinky-f4 -j 14 --
[ 16%] Built target CMSIS_LD_F407VG
[ 16%] Building C object CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj
[ 33%] Building C object CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c.obj
[ 50%] Building C object CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c.obj
[ 50%] Building C object CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c.obj
[ 66%] Building C object CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj
[ 66%] Building C object CMakeFiles/stm32-blinky-f4.dir/1b620444c1f354133dde702c2c9ed694/STM32Cube_FW_F4_V1.27.0/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c.obj
gcc.exe: warning: '-mcpu=' is deprecated; use '-mtune=' or '-march=' instead
gcc.exe: error: nosys.specs: No such file or directory
gcc.exe: warning: '-mcpu=' is deprecated; use '-mtune=' or '-march=' instead
gcc.exe: gcc.exe:error:  nosys.specs: No such file or directory
warning: '-mcpu=' is deprecated; use '-mtune=' or '-march=' instead
gcc.exe: warning:  gcc.exe:-mcpu= ' is deprecated; use 'error: -mtune=nosys.specs: No such file or directory
' or 'mingw32-make.exe[3]: *** [CMakeFiles\stm32-blinky-f4.dir\build.make:75: CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj] Error 1
-march=mingw32-make.exe[3]: *** Waiting for unfinished jobs....
' instead
gcc.exe:mingw32-make.exe[3]: *** [CMakeFiles\stm32-blinky-f4.dir\build.make:89: CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c.obj] Error 1
 mingw32-make.exe[3]: *** [CMakeFiles\stm32-blinky-f4.dir\build.make:103: CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c.obj] Error 1
warning: gcc.exe:' -mcpu=error: ' is deprecated; use 'nosys.specs: No such file or directory
gcc.exe:-mtune=' or ' -march=mingw32-make.exe[3]: *** [CMakeFiles\stm32-blinky-f4.dir\build.make:117: CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c.obj] Error 1
warning: ' instead
'-mcpu=' is deprecated; use 'gcc.exe:-mtune= ' or '-march=error: ' instead
nosys.specs: No such file or directory
gcc.exe:mingw32-make.exe[3]: *** [CMakeFiles\stm32-blinky-f4.dir\build.make:131: CMakeFiles/stm32-blinky-f4.dir/C_/Users/Mark/Downloads/en.stm32cubef4_v1-27-0/STM32Cube_FW_F4_V1.27.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj] Error 1
 error: nosys.specs: No such file or directory
mingw32-make.exe[3]: *** [CMakeFiles\stm32-blinky-f4.dir\build.make:158: CMakeFiles/stm32-blinky-f4.dir/1b620444c1f354133dde702c2c9ed694/STM32Cube_FW_F4_V1.27.0/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c.obj] Error 1
mingw32-make.exe[2]: *** [CMakeFiles\Makefile2:4107: CMakeFiles/stm32-blinky-f4.dir/all] Error 2
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:4114: CMakeFiles/stm32-blinky-f4.dir/rule] Error 2
mingw32-make.exe: *** [Makefile:2060: stm32-blinky-f4] Error 2
```

My understanding is, that it tries to compile using gcc, from mingw, and it fails to do so.

**What I've tried so far**
I've tried to modify `stm32_gcc.cmake` like this:
```
find_program(C_COMP NAMES ${STM32_TARGET_TRIPLET}-gcc HINTS ${TOOLCHAIN_BIN_PATH} REQUIRED)
find_program(CXX_COMP NAMES ${STM32_TARGET_TRIPLET}-g++ HINTS ${TOOLCHAIN_BIN_PATH} REQUIRED)
find_program(ASM_COMP NAMES ${STM32_TARGET_TRIPLET}-gcc HINTS ${TOOLCHAIN_BIN_PATH} REQUIRED)

set(CMAKE_C_COMPILER ${C_COMP})
set(CMAKE_CXX_COMPILER ${CXX_COMP})
set(CMAKE_ASM_COMPILER ${ASM_COMP})
```

And this prints correctly:
`Cxx compiler is : C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/11.2 2022.02/bin/arm-none-eabi-g++.exe`

The `CXX_COMP` variables were needed, because find_program didn't modify the original values for some reason, even with `REQUIRED`.

But even like this, the error message is still the same.

I would like to request an example, on how to compile on Windows, what generator/toolchain to use.

atsju, 2022-06-13:
Hi, on Vscode it should work quite easily. 
Install cmake-tools extension, ensure arm-nlne-eabi-gcc is in path and "scan for kits" With the extension. This way you will be able to generate for correct toolchain. 

Your explanation about failure cause are correct. The fix is related to path I think but not directly to thid repo.

MarkMan0, 2022-06-13:
Yes it does, my bad. Didn't have make installed, so I couldn't set the arm toolchain as kit. 
Maybe put *make* as a requirement to use this repo?

closed 2022-06-13 by MarkMan0

atsju, 2022-06-14:
Make isn't strictly required. You can use other generators like ninja build