ObKo/stm32-cmake #293: NO LUCK : Iam trying to build the project on windows 10

vinodneelakantam, 2022-07-09:
Hello Team , 

Iam trying to build this project on windows 10 . But , I am having difficulty in configuring the project . I believe documentation has enough verbosity . can someone help me to understand do i need to install any SDK for this ? 

i always end up getting error saying 

Detected toolchain path STM32_TOOLCHAIN_PATH in environmental variables: 
D:\Working\stm32-cmake\cmake\stm32_gcc.cmake
No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
The C compiler identification is GNU 9.2.0
The ASM compiler identification is GNU
Found assembler: C:/cip_tools/mingw/20200319/bin/gcc.exe
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working C compiler: C:/cip_tools/mingw/20200319/bin/gcc.exe - skipped
Detecting C compile features
Detecting C compile features - done
Search for CMSIS families: STM32F4
Search for CMSIS RTOS: RTOS;RTOS_V2
ENV STM32_CUBE_F4_PATH specified, using STM32_CUBE_F4_PATH: D:/Working/stm32-cmake/cmake/stm32
CMake Error at C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find CMSIS (missing: CMSIS_INCLUDE_DIRS CMSIS_SOURCES STM32F4)
  (found version "v1.1.0")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  D:/Working/stm32-cmake/cmake/FindCMSIS.cmake:321 (find_package_handle_standard_args)
  CMakeLists.txt:40 (find_package)

atsju, 2022-07-09:
Stm32cube_f4_path must be the cube clone path https://github.com/STMicroelectronics/STM32CubeF4

Also you will need Arm-none-eabi-gcc instead of mingw for compiler.

Hish15, 2023-01-02:
Hello @vinodneelakantam, this looks solved to me, I'm closing it.
If you have more question you can reopen or write a new issue.

BR

closed 2023-01-02 by Hish15