ObKo/stm32-cmake #296: Unable to configure/build Blinky example on windows 10. (No STM32_TARGET_TRIPLET specified)

mahanth07, 2022-07-20:
hello;

Actually I am using cmake GUI for configuration and building the Blinky from examples, but I am always ending up with this error:

**ERROR**
Detected toolchain path STM32_TOOLCHAIN_PATH in environmental variables: 
C:\MinGW\bin
No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
CMake Error at CMakeLists.txt:15 (project):
  Failed to run MSBuild command:

    MSBuild.exe

  to get the value of VCTargetsPath:

    The system cannot find the file specified

Configuring incomplete, errors occurred!
See also "C:/Users/DELL/Desktop/build/CMakeFiles/CMakeOutput.log".

mahanth07, 2022-07-20:
[image]

mahanth07, 2022-07-20:
I have installed it already!

mahanth07, 2022-07-20:
I have changed generator too, but ending with the same error. 
No I am using the cmake for the first time. Please could you help me in configuring/building this Blinky example.
Is there any other ways  to build by any command lines or other ide's?

atsju, 2022-07-20:
Personally on windows I have a preference for VScode. https://code.visualstudio.com/
Install the cmake-tools plugin.
open the folder stm32-cmake\examples\blinky in stm32-cmake\examples\blinky in VScode and let the IDE guide you. It should be smooth (arm-none-eabi-gcc must be in path)

Also you use a release candiddate for cmake. It's shouldn't be an issue but consider using a stable release :
https://cmake.org/download/

closed 2022-10-03 by Hish15