ObKo/stm32-cmake #297: CMake CMP0057 Policy Error

orangesalad, 2022-08-13:
**Describe the bug**
CMake issues a warning regarding policy CMP0057 being set to old by default. CMake also fails due to the use of `IN_LIST` in the `FindX.cmake` files. 

This affects the following files:
- `FindBSP.cmake`
- `FindCMSIS.cmake`
- `FindFreeRTOS.cmake`
- `FindHAL.cmake`

**Sources to reproduce**
 - [See here for an example](https://github.com/orangesalad/stm32-app)

**Steps to reproduce**
Perform `cmake ..` on the repository linked above.

**Environment:**
 - OS: Ubuntu 18.04.6 LTS
 - Compiler: arm-none-eabi-gcc 9.3.1 20200408
 - stm32-cmake: 1970eb1
 - cmake: 3.23.3
 - HAL/cube/CMSIS: 

**Additional context**
This issue is fixed by adding `cmake_policy(SET CMP0057 NEW)` to the affected files

orangesalad, 2022-08-13:
I actually realize now that this only occurs if you set `cmake_minimum_required` to be version 3.2 or less in the top level CMakeLists. The lib README clearly states CMake 3.16 or above so this is actually not an issue

Hish15, 2022-08-13:
As you noticed this issue is actually linked to a cmake version problem. While you can close this issue right away, you pointed out the fact that we don't check the cmake version. Adding a minimum version check with a message/warning seems mandatory to me.

closed 2022-08-13 by Hish15