ObKo/stm32-cmake #303: Support Dual Core Project WL55JCX

MoritzFranke, 2022-10-14:
Hello,
is there any chance I can get some support setting up a dual core project on the WL55JC1 chip?
greetings,
Moritz

atsju, 2022-10-14:
Hello Mortiz,
chance is very low that we make an example for this specific chip however we will be happy to add a "hello world" example if you provide it.
Best starting point for you might be the H7 example : https://github.com/ObKo/stm32-cmake/blob/master/examples/multi-core/CMakeLists.txt

MoritzFranke, 2022-10-14:
cmake_minimum_required(VERSION 3.16)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/../Tools/stm32-cmake-master/cmake/stm32_gcc.cmake)

project(stm32-template C ASM)

find_package(CMSIS COMPONENTS STM32WL_M4 STM32WL_M0PLUS REQUIRED)
find_package(HAL COMPONENTS STM32WL_M4 STM32WL_M0PLUS REQUIRED)

add_executable(m0core main.c)
add_executable(m4core main.c)

target_include_directories(m0core PRIVATE Inc)
target_include_directories(m4core PRIVATE Inc)

target_link_libraries(m0core PRIVATE
    HAL::STM32::WL::M0PLUS::RCC
    HAL::STM32::WL::M0PLUS::GPIO
    HAL::STM32::WL::M0PLUS::CORTEX
    CMSIS::STM32::WL::M0PLUS
    STM32::NoSys
)
stm32_print_size_of_target(m0core)

target_link_libraries(m4core PRIVATE
    HAL::STM32::WL::M4::RCC
    HAL::STM32::WL::M4::GPIO
    HAL::STM32::WL::M4::CORTEX
    CMSIS::STM32::WL::M4
    STM32::NoSys
)
stm32_print_size_of_target(m4core)

This is what i came up with based on that example, but i get errors:
[cmake] -- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
[cmake] -- Search for CMSIS families: STM32WL_M4;STM32WL_M0PLUS
[cmake] -- Search for CMSIS RTOS: RTOS;RTOS_V2
[cmake] CMake Error at C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
[cmake]   Could NOT find CMSIS (missing: CMSIS_INCLUDE_DIRS CMSIS_SOURCES STM32WL_M4
[cmake]   STM32WL_M0PLUS) (found version "v1.1.0")
[cmake] Call Stack (most recent call first):
[cmake]   C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
[cmake]   C:/Projects/LoRa/Tools/stm32-cmake-master/cmake/FindCMSIS.cmake:321 (find_package_handle_standard_args)
[cmake]   CMakeLists.txt:6 (find_package)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!

atsju, 2022-10-14:
seems correct at first glance but I have no time now to execute your code. 
In the meantime, could you please confirm the H7 example is working on your machine and also confirm you provided path to cube for cmake to search for files.

MoritzFranke, 2022-10-14:
Thx for editing
My best guess was that the PATH to CMSIS is wrong but could figure out where and how I can modify the PATH 

greetings

atsju, 2022-10-14:
I would recommend reading https://github.com/ObKo/stm32-cmake/blob/master/README.md#usage

And start by setting up your machine for the easiest example before going to dual-core:
https://github.com/ObKo/stm32-cmake/blob/master/examples/blinky/CMakeLists.txt

atsju, 2022-10-16:
@MoritzFranke did you fix your issue ? 
I just tried your cmakelist on my computer and generation works fine. 
You are likely missing `STM32_CUBE_<FAMILY>_PATH` as explained in README.

MoritzFranke, 2022-10-17:
Hello @atsju i havent been able to fix the porblem yet. I have made some progress, by trying the example dual core project. I managed to get the CMSIS imports right but then ran into issues with the HAL imports. Now I was trying the tests to ensure every Path is correct. But I could not figure out how the HAL works.
What is meant by the "<PathToCubeRoot>" ? 
Is it:
 - to the "STM32Cube" Folder (Contains: "Example" and "Rpository"
 - to the "STM32CubeIDE" Folder (Contains workspace)
 - to the "STM32Cube/STM32Cube_FW_<FAMILY>_V1.XX.X" Folder

Thanks for your Help.
Do you maybe have Discord for easier communication?

MoritzFranke, 2022-10-17:
I have a small addition, I get the error hint:
"note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code" when compiling, where do I specify this?

MoritzFranke, 2022-10-17:
I checked the env variables with "Get-ChildItem -Path Env:" and have the entry: "STM32_CUBE_WL_PATH             C:/Users/moritz/STM32Cube/Repository/STM32Cube_FW_WL/" but when I put "message("${STM32_CUBE_WL_PATH}")" inside my CMakeList.txt file I get the result: "[cmake] /opt/STM32CubeWL" 
Do you know why that is?

MoritzFranke, 2022-10-17:
I fixed the Issue by adding this loop to the test file:
"
"
foreach(FAMILY ${TEST_FAMILIES})
    set(STM32_CUBE_${FAMILY}_PATH C:/Users/moritz/STM32Cube/Repository/STM32Cube_FW_${FAMILY})
    message("THE PATH TO ${FAMILY} is ${STM32_CUBE_${FAMILY}_PATH}")
endforeach()
"
now the HAL test works fine.
I dont know why my env variables wont be added permanently and why they could not be accessed by the cmake files but its a workaround that is ok for me

atsju, 2022-10-17:
Please allow me to close issue as it works for you. What you did seems fine.

Have you restarted vscode after changing windows path ? if not it's normal that the new env variable is not taken into account.

closed 2022-10-17 by atsju