ObKo/stm32-cmake #305: Add Branch with only CMAKE functions called `lib `

simoneruffini, 2022-12-07:
Create a stripped down version of the master branch called `lib` that contains only the CMAKE functionality (the content of the `cmake` folder). This branch could be called `lib` and can be used directly for git-submodules inside stm32 projects.
The benefit of this branch is less imported code in a new project.
The branch can be automatically generated from `master` by a github action, it will live on its own, never merged into any branch but only merging/rebasing  with `master`.

simoneruffini, 2023-01-12:
This has the problem that I the repo can't be added as a git-submodule that helps with keeping track of remote updates and changes.

> Do you now any library that does the thing you describe?
I don't know... I use it, although for me is manually performed and not with a github action.

atsju, 2023-01-12:
If we group necessary file in a "lib" folder, you should be able to add repo as submodule and only build/import this folder while ignoring other files.  
I'm not sure creating a special branch here is the correct answer when build system is able to take care of it. Or am I missing something ?

simoneruffini, 2023-01-13:
I can already do that with the current directory structure, I just didn't want to have all the examples in the repo too. Sometimes the build system is just an "include all *.c *.h" files and this could mess up stuff, but yes ok maybe is just lazyness

Hish15, 2023-01-13:
Hi @simoneruffini, I'm closing this issue.
I advice against globing all c file from a directory. This is not the way to go on modern Cmake :) and as you mention, is not clean since it can bring unwanted things to the build list!

closed 2023-01-13 by Hish15