ObKo/stm32-cmake #306: FetchContent_Declare does not show progress

julian-poidevin-switchess, 2022-12-19:
**Describe the bug**
When using the FetchContent features of this repository, the `git clone` progress is supposed to show in the terminal but it does not show.

**Sources to reproduce**
 - Use one of the example provided by this repository and make the target

**Expected behavior**
The git clone progress should be shown.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Compiler: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
 - stm32-cmake: 6d07cb82bce578369984533a755908ea38b43b19
 - cmake: `3.22.1`
 - HAL/cube/CMSIS: N/A

**Additional context**
The problem can be solved by setting : `FETCHCONTENT_QUIET` to `FALSE` : `set(FETCHCONTENT_QUIET FALSE)`. See https://cmake.org/cmake/help/latest/module/FetchContent.html#variable:FETCHCONTENT_QUIET

Hish15, 2022-12-20:
Hello @julian-poidevin-switchess,

Cmake's documentation under your link specifies that `FETCHCONTENT_QUIET` must not be set to OFF by the project but by the final user. This is done to reduce the noise on the configuration stage.

I suggest we follow this recommendation and let the user set it if needed. We can add something like this on the examples :
```cmake
#set(FETCHCONTENT_QUIET OFF) #Unomment this line to see logs and progress bar on the fetched content
```

julian-poidevin-switchess, 2022-12-20:
Hi @Hish15, thanks for the fast feedback !

Yeah, that would be great to add the recommandation in the examples and also maybe update the project documentation to specify that this option exists !

Hish15, 2022-12-20:
You're welcome!

I don't think we want to mirror all cmake documentation on the already heavy documentation we have on this repo...
That being said some aspects of cmake can be/are documented here. The one you mention might be "useless" since with the default value  `ON` you still get error messages in case of error. 
I'm not 100% against adding it to this repo documentation. Only 80% 😄 
I will let my peer @atsju give it's 2 cents on the matter.

atsju, 2022-12-20:
Hi,
I'm 100% aligned with @Hish15 analysis. 

1. This option must be default `ON` 
2. We cannot add each and every CMAKE related option in our examples and documentation

I would vote against documenting it (it is 100% cmake thing).  
Also against adding it to example as it's doesn't help any diagnostic (errors are still shown without).

Hish15, 2022-12-20:
Thank's @atsju.

@julian-poidevin-switchess I'm closing this issue with no changes on the repo.

Have a nice day

closed 2022-12-20 by Hish15