ObKo/stm32-cmake #307: Wrong HAL family in find_package

Dooez, 2022-12-31:
**Describe the bug**
When building from a clean project and fetching multiple HAL families CMake Error happens. For example, when trying to build fetch-cmsis-hal example:

```
CMake Error at /<path...>/stm32-cmake/cmake/FindHAL.cmake:79 (message):
  could not find HAL for family F4
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)
```
If i add following code:
```
# Step 2 : Generating all the valid drivers from requested families
...
# added portion
    message("${FAMILY}")
    message("${FAMILY_L}")
# added portion over

    find_path(HAL_${FAMILY}_PATH
```
The output from it is:
```
F4
l0
```

After i added FAMILY_L assignment the problem was solved
```
    if(CMAKE_MATCH_1) #Matches the family part of the provided STM32<FAMILY>[..] component
        set(FAMILY ${CMAKE_MATCH_1})
        string(TOLOWER ${FAMILY} FAMILY_L) # my addition
    endif()
```

**Sources to reproduce**
The default fetch-cmsis-hal example is enough to reproduce

**Steps to reproduce**
1. Go to '/.../stm32-cmake/examples/fetch-cmsis-hal'
2. Type 'cmake .'
3. See error

**Expected behavior**
Successful build.

**Environment (please complete the following information):**
 - OS: Ubuntu-22.04 in WSL2
 - Compiler: arm-none-eabi-gcc 10.3.1
 - stm32-cmake: 6d07cb82
 - cmake: 3.22.1
 - HAL/cube/CMSIS: auto fetch

Hish15, 2022-12-31:
I was able to reproduce. Thank you for your feed back.  
We have testd on the fetch feature here https://github.com/ObKo/stm32-cmake/tree/master/tests/cmsis.
I think it does make sense to make github action run the provided examples too. Those examples are indeed like tests : they should never break!

Hish15, 2023-01-01:
Hi @Dooez,
Since you found a solution to make it work do not hesitate to push a PR to this repo : we will gladly merge it
I will dig into the tests issue afterwards

PR #308 (2023-01-01): Set lowercase family string in step 2 of find_package(HAL ...);
This pull request solves issue #307 for me;
I am not experienced with complex CMake and this particular toolchain, but hopefully, this is a generic fix.

closes #307
closed 2023-01-01 by Hish15