ObKo/stm32-cmake #311: quotes in --sysroot breaks clangd

Logiase, 2023-01-18:
**Describe the bug**
set `--sysroot` in `utilities.cmake` breaks clangd.

**Sources to reproduce**
any exmaple.

**Steps to reproduce**
1. Copy and build an example
2. take a look at `compile_commands.json`, and you will see `--sysroot=\\\"/usr/arm-none-eabi\\\"`
3. run `clangd --check={any source}`, and there's an error means clangd cant find system headers, and check `interlnal (cc1) args` in output, the redundant quotes and slash breaks the path.

**Expected behavior**
clangd is a very common lsp server, it should works fine

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**
 - OS: ArchLinux
 - Compiler: arm-none-eabi-gcc 12.2.0
 - stm32-cmake: abea01902bead78d9a5dd2f7603937da121fcc2c (newest)
 - cmake: 3.25.1

 - HAL/cube/CMSIS: stm32cubeh7 v1.11.0

**Additional context**
This problem has no effected on compilation, but it cause most IDE or editors which use clangd as LSP server does not work.

For example, `math.h` not found.

Hish15, 2023-01-18:
Hello,

Thanks for this detailed issue. I can confirm this is an issue right now, but the solution is actually already on PR #290.
I will try to merge this PR and solve your issue :) 

Cya

Logiase, 2023-01-19:
> 

Hi

PR #290 can fix this issue, and correctly handles the spaces in sysroot paath.

with spaces:
[image]

without spaces:
[image]

atsju, 2023-01-19:
closing as fixed in #290

closed 2023-01-19 by atsju