ObKo/stm32-cmake #316: Multiple warnings from HAL

Posejdos, 2023-03-06:
When doing clean builds of my projects, I get several warnings coming from HAL libraries.
Is there any known way to disable them? I tried adding SYSTEM to `target_include_directories(...)` in `FindHAL.cmake` but it didn't work.

I really don't want to modify the HAL libraries.

atsju, 2023-03-06:
Hi, warnings is not something you want to normalize, ignore or disable in general except if you know exactly what you are doing. 

Maybe you want to share the warning you are facing ? and/or a minimal code to reproduce the issue ? We won't be able to help if we can't reproduce.

Posejdos, 2023-03-06:
Hi, thanks for the quick answer!

I know I shouldn't ignore my own warnings, but I don't want to be told about warnings in libraries. Especially when they are mostly about unused variables. I also very much don't want to edit them in any way (to get rid of warnings), because that's just bad practice - what happens when the libraries get updated?

Sorry, I should've listed the steps to reproduce in the first place. Here they are:

1. Clone this repo
2. Set your environment variables if needed to allow compilation
3. Go to _examples/blinky_
4. Add the PWR HAL module  in _CMakeLists.txt_:

`set(HAL_COMP_LIST RCC GPIO CORTEX PWR)`

`target_link_libraries(stm32-blinky-f4
        HAL::STM32::F4::RCC
        HAL::STM32::F4::GPIO
        HAL::STM32::F4::CORTEX
        HAL::STM32::F4::PWR
        CMSIS::STM32::F407VG
        STM32::NoSys)`

Posejdos, 2023-03-06:
Hi, thanks for the quick answer!

I know I shouldn't ignore my own warnings, but I don't want to be told about warnings in libraries. Especially when they are mostly about unused variables. I also very much don't want to edit them in any way (to get rid of warnings), because that's just bad practice - what happens when the libraries get updated?

Sorry, I should've listed the steps to reproduce in the first place. Here they are:

1. Clone this repo
2. Set your environment variables if needed to allow compilation
3. Go to _examples/blinky_
4. Add the PWR HAL module  in _CMakeLists.txt_:

`set(HAL_COMP_LIST RCC GPIO CORTEX PWR)`

`target_link_libraries(stm32-blinky-f4
        HAL::STM32::F4::RCC
        HAL::STM32::F4::GPIO
        HAL::STM32::F4::CORTEX
        HAL::STM32::F4::PWR
        CMSIS::STM32::F407VG
        STM32::NoSys)`

5. Edit _cmake/stm32/f4.cmake_ to enable all warnings:
`target_compile_options(STM32::F4 INTERFACE 
    -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard
    -Wall -Wextra -Wundef
)`

6. Build the blinky example and you'll see a warning about an unused parameter.

In my projects I have more modules and they emit more warnings - this clutters my build output and makes it difficult to fix my own warnings. I'm pretty sure it should be possible to flag those libraries as third-party (system?) and disable these warnings for them, I just don't know how to do it.

Sorry for the closing and reopening the issue BTW. Missclick.

atsju, 2023-03-06:
maybe something like `target_compile_options(stm32-blinky-f4 -Wall -Wextra -Wundef)` to add the options only on the target you need them instead to lowest target and propagate? This is what targets are for.

Sorry I have not been able to try out the code but maybe this will unlock you in the meantime.

Posejdos, 2023-03-06:
Removing the warning options from _f4.cmake_ and adding them in the project CMakeLists.txt with
`target_compile_options(stm32-blinky-f4 PRIVATE -Wall -Wextra -Wundef)` makes no difference, the warning from HAL still shows, unfortunately.

There's no time pressure, so don't worry - I have been struggling with this for some time now, I'm used to it :) Take it easy.

atsju, 2023-03-06:
One ugly solution is to add property specifically to your sources `set_source_files_properties(blinky.c PROPERTIES COMPILE_FLAGS -Wall -Wundef -Wextra)`

------

I tried to compile HAL as a static library and then link it against blinky while adding compile options only to blinky but this also let the compile options down to hal... So following did not work :
```cmake
    add_executable(stm32-blinky-f4 ${MAIN_SOURCE_FILE} )
    
    target_link_libraries(stm32-blinky-f4 PRIVATE hal_lib)

    add_library(hal_lib STATIC stm32f4xx_hal_conf.h)
    target_link_libraries(hal_lib PUBLIC
        HAL::STM32::F4::RCC
        HAL::STM32::F4::GPIO
        HAL::STM32::F4::CORTEX
        HAL::STM32::F4::PWR
        CMSIS::STM32::F407VG
        STM32::NoSys
    )
    target_compile_options(stm32-blinky-f4 PRIVATE  -Wall -Wundef -Wextra)
    
    stm32_print_size_of_target(stm32-blinky-f4)
```

Posejdos, 2023-03-07:
Oh my, it works! Thank you kindly. My cmake-fu wasn't strong enough to find this solution by myself.
I'm closing this issue - thanks again!

closed 2023-03-07 by Posejdos