ObKo/stm32-cmake #319: How to include middleware?

Posejdos, 2023-04-20:
Hi,
I'm using STM32F042 for a USB application and want to include proper middleware to handle the USB functionality. More specifically, I'm talking about the **STM32_USB_Device_Library** found in the Cube Firmware Package (in my case, it's **STM32CubeF0_V1.11.4**.

I found #65 which was merged, but the file committed there is no longer in the master branch. I copied it and tried to include and use it in my project, but I didn't have any success. It's probably due to my poor CMake skills.

I also didn't find anything in the README or issues. Could you please point me in the right direction?

atsju, 2023-04-20:
Hi, 
indeed, we are not handling every libraries supported by STM32. We support mainly HAL, LL and maybe some special libraries that were easy to add. 

What you need to do here, is create a library for USB sources, then add sources and include directories and then link you STM32 application to your library.

```cmake
add_library(my_USB_lib)
target_include_directories(my_USB_lib PUBLIC usb_dirs)
target_sources(my_USB_lib PRIVATE USB_sources.c)
target_link_libraries(my_USB_lib every cmake lib you need for USB to work)
```

Posejdos, 2023-04-20:
Thank you for your fast response.

I tried what you suggested and did some progress. The USB library tried to compile and stopped with an error message that made me see that it actually depends on _usb_device.h_, which depends on HAL, which is seen by CMake as another library in my project and I really didn't want to try to make it work. My application also depends on _usb_device.h_. 

Seeing that this is very convoluted for me I decided to stop trying to use this library as a CMake library and just add its sources directly in my application.
For future reference or anyone stumbling upon this issue, here is what I did:

```
set(SOURCES
    # My app sources here

    # These files are needed by my application. This is how STM32CubeIDE added them when I included USB in a test project
    USB/App/usb_device.c
    USB/App/usbd_cdc_if.c
    USB/App/usbd_desc.c
    USB/Target/usbd_conf.c

    ${STM32_CUBE_F0_PATH}/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c
    ${STM32_CUBE_F0_PATH}/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c
    ${STM32_CUBE_F0_PATH}/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ioreq.c
    ${STM32_CUBE_F0_PATH}/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c
)

add_compile_definitions(STM32F042x6)
add_executable(MY_PROJECT ${SOURCES} stm32f0xx_hal_conf.h)
target_include_directories(MY_PROJECT PUBLIC 
    USB/App/ USB/Target/ 
    ${STM32_CUBE_F0_PATH}/Middlewares/ST/STM32_USB_Device_Library/Core/Inc/ 
    ${STM32_CUBE_F0_PATH}/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc/
)
```
 This seems to work. Thank you for help. Feel free to close this issue if you don't have anything to add.

atsju, 2023-04-20:
Thank you for feedback. This is indeed easiest solution and is more than acceptable for small applications.

closed 2023-04-20 by atsju