ObKo/stm32-cmake #321: How to include CMSIS arm_math?

Posejdos, 2023-05-10:
Hi,
I need to use the `arm_math` library in my application, but I don't see a way to include anything else than the device headers from CMSIS.

I can see CMSIS is included in `${STM32_CUBE_XX_PATH}/Drivers/CMSIS/DSP/Include` - but all I can find there are headers and object files. So I think I have to include the headers and link them to the object files. I have no idea how to make this magic work.

The use case is: I use an external library ([https://github.com/drowe67/codec2](url)). I include it in my project like this:
```
...flags and stuff...
# add_definitions(-DFDV_ARM_MATH)
add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/codec2 EXCLUDE_FROM_ALL)
target_compile_options(codec2 PUBLIC
    -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard
    -ffunction-sections -fdata-sections -Os)
```

```
target_link_libraries(MyApp
    ...HAL and stuff...
    codec2
)
```

This works great. Now for the sake of optimization I'd like to enable the `FDV_ARM_MATH` flag which enables `arm_math.h` usage. Now it won't compile: `fatal error: arm_math.h: No such file or directory`.

Thanks in advance.

Hish15, 2023-05-11:
Hello,

For now the DSP part of CMSIS is not handled. To make it work, you must create a target on your project and add everything missing there.
 - Starting with Drivers/CMSIS/DSP/Include
 - Some C file from Drivers/CMSIS/DSP/Source probably
 - More ?

Hish15, 2023-05-11:
I'm willing to help you out writing the target, but I will needed a repository that does reproduce the problem you have right now :)

Posejdos, 2023-05-11:
I managed to create a minimal repo that allows to reproduce my issue:
[https://github.com/Posejdos/cmake_stuff](url)

You'll see in `CMakeLists.txt` that I managed to build and use the DSP library in a pretty simple (and probably not right) manner.

Now, when you uncomment line 48 in `CMakeLists.txt` (adding the `FDV_ARM_MATH` flag) the `codec2` library will try to include the `arm_math.h` header and use it. What can I do to make it work?

Thank you for time.

Hish15, 2023-05-12:
Hello, I tried to build the repo you linked and run into 2 problems (most probably linked)
  1) The submodule codec2 can not be restored (the SHA does not exist) I set the submodule to origin/master (f68a2fe4)
  2)  Posejdos/cmake_stuff/codec2/src/codec2.h:32:10: fatal error: codec2/version.h: No such file or directory
   32 | #include <codec2/version.h>

Posejdos, 2023-05-13:
I'm so sorry! I didn't check if it works on a clean git pull :) My bad.

I did a very ugly workaround, but it's OK for this temporary repo. Everything should build now.

Hish15, 2023-05-14:
Hello,

I had a look at your repo, and could reproduce your problem.
The thing is you create a DSP target but does not link it to the codec2 target.
linking the codec2 and DSP to the same top executable is not enough.

That being said doing it properly open the realm of other problems linked to the fact that codec2.
`CMake Error: install(EXPORT "codec2-config" ...) includes target "codec2" which requires target "DSP" that is not in any export set.`
I'm not used to exporting targets and the install command so I cannot help you without a deep dive into it. 

Ps: there is a folder stm32 in the codec2 folder!!!

Good luck

Posejdos, 2023-05-16:
OK, thank you for your time.
I'll investigate the exporting issue. I feel like it's a rather simple problem I'm having, yet I'm lacking the skills to solve it...

From what I can see the stm32 subproject doesn't use the `arm_math.h`, unfortunately.

Hish15, 2023-05-16:
This repo does not use it, but if it did it would be done as you did in your test repo...
Good luck

closed 2023-05-16 by Hish15