ObKo/stm32-cmake #326: Calling find_package() to find CMSIS multiple times leads to errors

Joaliasbarb, 2023-06-29:
**Describe the bug**
CMake errors are popping up when calling find_package multiple times.
```
CMake Error at C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/cmake/FindCMSIS.cmake:107 (add_custom_target):
  add_custom_target cannot create target "CMSIS_LD_F401CB" because another
  target with the same name already exists.  The existing target is a custom
  target created in source directory
  "C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/examples/blinky".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/cmake/FindCMSIS.cmake:268 (cmsis_generate_default_linker_script)
  CMakeLists.txt:41 (find_package)

CMake Error at C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/cmake/FindCMSIS.cmake:266 (add_library):
  add_library cannot create imported target "CMSIS::STM32::F401CC" because
  another target with the same name already exists.
Call Stack (most recent call first):
  CMakeLists.txt:41 (find_package)

CMake Error at C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/cmake/FindCMSIS.cmake:107 (add_custom_target):
  add_custom_target cannot create target "CMSIS_LD_F401CC" because another
  target with the same name already exists.  The existing target is a custom
  target created in source directory
  "C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/examples/blinky".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  C:/heig-vd/projects/stm32-cmake/stm32-cmake-multiple-find-package-calls/cmake/FindCMSIS.cmake:268 (cmsis_generate_default_linker_script)
  CMakeLists.txt:41 (find_package)
```

**Sources to reproduce**
Please provide link to repository showing the problem.
[link to the commit causing an issue](https://github.com/Joaliasbarb/stm32-cmake-multiple-find-package-calls/commit/f86ab2f4c679a64246f7bb0723ad6b075e01c744)

**Steps to reproduce**
1. Add multiple calls to the find_package function for CMSIS.

**Expected behavior**
On the first find_package call, the behavior shall not change. On subsequent calls, already existing custom targets shall not be created again.

**Environment :**
 - OS: Windows
 - Compiler: arm-none-eabi-gcc (Arm GNU Toolchain 12.2.MPACBTI-Rel1 (Build arm-12-mpacbti.34)) 12.2.1 20230214
 - stm32-cmake: v2.1.0
 - cmake: 3.25.0
 - HAL/cube/CMSIS: stm32cubeF4 v1.27.1

Joaliasbarb, 2023-06-29:
[Here's](https://github.com/Joaliasbarb/stm32-cmake-multiple-find-package-calls/commit/4a2aa96d4cad7e9f939af221a0692476b9d637de) a fix proposition (not sure if this is correct though).

Hish15, 2023-06-29:
It's weird to call the find_package command twice. We can hide the error but consider this : 

```cmake
find_package(CMSIS v1.0.0 COMPONENTS STM32F4 REQUIRED)
.
.
.
#With you proposal this will fail silently (not retrieving the v2.0.0 version)
find_package(CMSIS v2.0.0 COMPONENTS STM32F4 REQUIRED) 
```

Joaliasbarb, 2023-06-29:
Of course, if we look at the modified example, this may seem weird to call find_package twice. I stumbled upon this error because I have a repo for a project where multiple executables are compiled and sometime, for the same platform. For each executable, I would like to find the required CMSIS + HAL components.

The error is actually located in the cmsis_generate_default_linker_script function. Maybe it would be better to place the if statement in there to avoid creating the custom targets multiple time.

Hish15, 2023-06-29:
Ok, do open a PR so we can see :)

nische, 2024-06-03:
Hi, 
did you have any news here?

nische, 2024-06-05:
I open a pull request with a little fix. That works for me at the moment but i dont know if anything other is broken now :-D 

(Think i make something wrong while linking of pullrequest and issue) 

Pullrequest: #343