ObKo/stm32-cmake #350: Custom linker script example not working

jaypacamarra, 2025-01-23:
**Describe the bug**
I am trying to pass in a custom linker script to my build using the function `stm32_add_linker_script(stm32-blinky-f4 PRIVATE F407VG.ld)` but the elf that is generated does not have correct load and execute address in flash.  

**Sources to reproduce**
[https://github.com/jaypacamarra/stm32-cmake](https://github.com/jaypacamarra/stm32-cmake)

**Steps to reproduce**
1. Go to blinky example in `examples/blinky`
2. build `BLINKY_F4_EXAMPLE`
3. run `readelf -l build/stm32-blinky-f4`

**Expected behavior**
I am expecting FLASH to start at `0x08020000` but instead it is still `0x08000000`

**Screenshots**
```
> readelf -l build/stm32-blinky-f4.elf 

Elf file type is EXEC (Executable file)
Entry point 0x80008ad
There are 3 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x001000 0x08000000 0x08000000 0x00a68 0x00a68 RWE 0x1000
  LOAD           0x002000 0x20000000 0x08000a68 0x00010 0x00010 RW  0x1000
  LOAD           0x000010 0x20000010 0x20000010 0x00000 0x00db8 RW  0x100

 Section to Segment mapping:
  Segment Sections...
   00     .isr_vector .text .init_array .fini_array 
   01     .data 
   02     .bss ._user_heap_stack .bss ._user_heap_stack 
```

**Environment (please complete the following information):**
 - OS: `Arch Linux`
 - Compiler: `arm-none-eabi-gcc v14.2.0`
 - stm32-cmake: [5f5c92c](https://github.com/ObKo/stm32-cmake/commit/5f5c92c1a0a88dd5d214f60402d4244af4e3b795)
 - cmake: `3.31.4`
 - HAL/cube/CMSIS: `stm32cubef4 v1.28.0`

**Additional context**
I'm just implementing and testing out stm32-cmake for a bootloader project so I need to offset my application in flash.

Hish15, 2025-01-23:
Hello,

You do have a custom linker script example, here you started from the blinky example.

In the readme :
```
So, if you don't need linker script, you can link only CMSIS::STM32::<TYPE> library and provide your own script using stm32_add_linker_script function
```

Try changing this line, to link the family instead and you will be good to go

https://github.com/jaypacamarra/stm32-cmake/blob/efebc9d5daccfc4c5ca49c4503032550b6eeae55/examples/blinky/CMakeLists.txt#L59

Hish15, 2025-01-23:
Ok I see what you meant : the linker-script example does not link the CMSIS::STM32::<TYPE> therefore does not work.

Can you check that following the readme does fix it, and open a PR to so we can clean it please ?

jaypacamarra, 2025-01-23:
awesome I get the proper elf now when i switched F407VG to F407xx. Thanks.

closed 2025-01-23 by jaypacamarra