ObKo/stm32-cmake #351: Add cmsis-core version variable

cosmikwolf, 2025-03-06:
CMSIS core is locked to verison v5.6.0 in cmake/stm32/devices.cmake, but 5.9.0 has been released

There should be a variable that allows the user to set the CMSIS core version

Hish15, 2025-03-12:
Hello @cosmikwolf, your idea makes sense, we could introduce
```
set(STM32_CMSIS_VERSION 5.9.0 CACHE STRING "Version of cmsis_core to fetch when using the stm32_fetch_cmsis function")
```

You can do one two things : 

- Open a PR to change this one line https://github.com/ObKo/stm32-cmake/blob/5f5c92c1a0a88dd5d214f60402d4244af4e3b795/cmake/stm32/devices.cmake#L169
- Introduce the proposed variable, and document it, more lines, but better solution.

ashen1queter, 2025-04-12:
Hey there, created a pr(#352 ) regarding the above.